halfmarble/hm-panelizer #10: Outline verification incorrect

jamosaur, 2022-09-09:
**Describe the bug**

With small PCB's, the outline verification is incorrect and causes the panelised output to have no edgecuts whatsoever.

as you can see in the screenshots, the layout validation is a fraction of the size of the PCB.

**To Reproduce**
Steps to reproduce the behavior:

import gerbers, set JLC type, set panel to 8x5 (breaks even at 2x1)

**Expected behavior**
A panelised PCB to be created

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux

robots, 2023-08-25:
Hi,
wanted to use this project to panelize files from eagle. But i think i have found the same problem :)

[image]

PCB verification outline is scaled a bit, and wont align with PCB.

gerbv rendering is correct. I dont know where is the problem. Maybe some hints on where to look in the code ? Same output is from pcb-tools, so i think the underlying project might be the problem.

robots, 2023-08-25:
Hi,

I dont know if this is relevant to @jamosaur , but here is fix for "my" problem.

```py
diff --git a/Pcb.py b/Pcb.py
index b4d5c81..709e5da 100644
--- a/Pcb.py
+++ b/Pcb.py
@@ -77,7 +77,7 @@ class PcbPrimitive:

 class PcbOutline:

-    def __init__(self, path, size, verbose=False):
+    def __init__(self, path, size, verbose=True):

         self._valid = True

@@ -137,7 +137,7 @@ class PcbOutline:
                         good = y_delta == 0.0
                         self._horizontal.append(PcbPrimitive(line, edge, good))

-            self._scale = size / max(self._max_x, self._max_y)
+            self._scale = size / max(self._max_x-self._min_x, self._max_y-self._min_y)
         else:
             self._min_x = 0.0
             self._min_y = 0.0

```

My gerber's origin is not at 0,0 but slightly offset. The max_x and max_y needs to be offset by this _min_x/y otherwise the scale will get scewed.